emlid/Navio-SDK-Windows-IoT #1: Deploy Hardware Project to NuGet

Narnain, 2015-08-17:
It would be useful for the Hardware project to be available in NuGet.  This will make it easier to pull in the latest code to custom flight control software as this project progresses.

CodeChief, 2015-08-23:
Totally agree. This should occur latest when the Core SDK is finished but would be useful sooner. NuGet is much better than downloading and managing references to third party APIs manually.

With the release of Visual Studio 2015 it's interesting to note a curious change of direction at Microsoft. They are now promoting other package managers for their cross-platform solutions, but so far I guess this is intended to improve their web and JavaScript projects.

So before integrating NuGet (which I also prefer and hope is still recommended) it will be necessary to double-check what's happening here, i.e. which system Microsoft really want us developers to use in the future with their Universal Application Platform.

Narnain, 2015-08-24:
Microsoft are still deploying all their components via NuGet.  The addition of NodeJS has allowed Microsoft to support other package managers such as bower.

I think I saw that vNext library projects will build as NuGet packages by default, so I think betting on NuGet is still valid.

CodeChief, 2015-08-27:
Sorry my mistake closing issue, so re-opened now (the button is grey like a normal cancel button on Github Web site, should be red with confirmation!). So to explain recent updates I've gone through and added issues, milestones and labels for all the work items. Then users can better track progress using the built-in Github functionality.

CodeChief, 2015-10-07:
This has been implemented in the v1.0.4-Alpha release (just checked-in). To get the package search for the ID "Emlid.WindowsIoT.Hardware" with the "pre-release" option enabled. Enjoy!

closed 2015-10-07 by CodeChief