emlid/Navio-SDK-Windows-IoT #14: Build error .pfx file

funreadsapp, 2016-11-09:
Hello, 

i am trying to build the project in VS 2015 but i get build errors saying:

Error		Importing key file "Navio2Baro.pfx" was canceled.	Emlid.WindowsIot.Hardware			
Error		Cannot import the following key file: Navio2Baro.pfx. The key file may be password protected. To correct this, try to import the certificate again or manually install the certificate to the Strong Name CSP with the following key container name: VS_KEY_1CA9B8514D25F494	Emlid.WindowsIot.Hardware			
Error		Importing key file "..\..\..\Common\TemporaryKey.pfx" was canceled.	Navio Barometer CS Sample			
Error		Cannot import the following key file: ..\..\..\Common\TemporaryKey.pfx. The key file may be password protected. To correct this, try to import the certificate again or manually install the certificate to the Strong Name CSP with the following key container name: VS_KEY_534360A408701E7F

i was wondering if this is a issue with my settings? or there is a password that doesn't come in correctly?
i tried the 
    cd Tools\Scripts
   PowerShell.exe -File "Setup Local.ps1"
    exit

but i got execution policy error, so i used powershell instead and didn't get an error, but i am still unable to build.

any help on this would be great
Thank you

CodeChief, 2016-11-11:
This is the normal message which is documented. It is telling you the VS_KEY to use as the parameter to the "Common\Temporary Key Install.cmd", i.e. you should open a command prompt from Visual Studio in the Common directory and enter: "Temporary Key Install.cmd" VS_KEY_534360A408701E7F

Then it will work. I suppose this is all over the place, and what we should do is move everything to a Setup.cmd in the solution root rather than hidden away in subdirectories. I'll make a note of that, I'm planning to spend more time on this project again now.

CodeChief, 2016-11-11:
I just tested and noticed that in addition the temporary key has expired. I've generated a new one and committed the changes. So you should be able to do a successful build.

For easy access to the command prompt from Visual Studio I recommend the "[Productivity Power Tools 2015](https://visualstudiogallery.msdn.microsoft.com/34ebc6a2-2777-421d-8914-e29c1dfa7f5d)" extension, then you can just right-click the Solution Explorer (e.g. root or a project) choose "Power Commands -> Open Command Prompt".

closed 2016-11-11 by CodeChief

funreadsapp, 2016-11-14:
Thank you, i followed you instructions and was able to get the software onto the pi.

i was just curious though. any idea when gps and imu will make it onto the project?
thanks again

CodeChief, 2016-11-14:
Good to hear. It will probably take 1-2 weeks to implement each chip.

microcompiler, 2016-11-15:
I have forked this project and completed a rough version of both the IMU (Mpu9250) and the GPS(NEO-M8N) devices.  The basic drivers are working for both chips.   The IMU has math/scaling issues that I need to figure out and the GPS needs some refactoring but in general both are working prototypes.

CodeChief, 2016-11-16:
This issue is closed. Please refer to the Emlid forum for futher comments and general updates: https://community.emlid.com/t/navio-sdk-for-windows-iot/381/59